vapourismo/knx-go #1: example

reejinbouk, 2017-06-19:
Hello,
in example 
// Send 20.5°C to group 1/2/3.
	err = client.Send(0, 2563, dpt.ValueTemp(20.5).Pack())
	

How to convert address "1/2/3" to uint (2563)?
Can we use another DPTs (not only for temperature)?

vapourismo, 2017-06-19:
I haven't added address generation yet. I'll add that soon. For now, keep in mind the generation works like this: `(a << 11) | (b << 8) | c` for address `a/b/c`.

You can use every DPT there is. Again, not implemented yet. Only Switch and ValueTemp are there for testing purposes.

The Send method takes a `[]byte` as parameter, if you know how to generate the application data unit, you can already use it.

vapourismo, 2017-06-19:
fba731901768fc7f674d08d0aa014b58a5de87c5 has support for address generation. See [NewGroupAddr3](https://godoc.org/github.com/vapourismo/knx-go/knx/cemi#NewGroupAddr3) and [NewGroupAddr2](https://godoc.org/github.com/vapourismo/knx-go/knx/cemi#NewGroupAddr2) in the `cemi` package.

closed 2017-06-19 by vapourismo