vapourismo/knx-go #2: Get Write, Read and Response messages

telemac, 2017-07-02:
Hello,
Thank you very much for sharing your great work on knx in go !

i wanted to get the cemi.GroupValueRead events, so i modified groups.go this way :

// GroupComm represents a group communication.
type GroupComm struct {
	Source      cemi.IndividualAddr
	Destination cemi.GroupAddr
	Data        []byte
	Command     cemi.APCI
}
...
			if app, ok := ind.Data.(*cemi.AppData); ok && (app.Command == cemi.GroupValueResponse || app.Command == cemi.GroupValueWrite || app.Command == cemi.GroupValueRead) {
				outbound <- GroupComm{
					Source:      ind.Source,
					Destination: cemi.GroupAddr(ind.Destination),
					Data:        app.Data,
					Command:     app.Command,
				}
			} else {
				util.Log(inbound, "Received L_Data.ind frame does not contain application data")
			}

I'm really new to your code and don't know if it has negative side effects.

Is it possible to get more events, on may be to have a subscription mechanism ?
Regards
Alexandre

vapourismo, 2017-07-02:
Thanks!

It should work. But I don't recommend doing it this way.

`GroupComm` is meant for group messages that carry data. `GroupValueRead`s don't have any payload.

Also the expression `app.Command == cemi.GroupValueResponse || app.Command == cemi.GroupValueWrite || app.Command == cemi.GroupValueRead` is basically always true, so you might omit that.

I recommend having a look at [Tunnel](https://godoc.org/github.com/vapourismo/knx-go/knx#Tunnel) or [Router](https://godoc.org/github.com/vapourismo/knx-go/knx#Router). If you check whether the [Message](https://godoc.org/github.com/vapourismo/knx-go/knx/cemi#Message) that is yielded by the [Inbound](https://godoc.org/github.com/vapourismo/knx-go/knx#Router.Inbound) channel is a [LDataInd](https://godoc.org/github.com/vapourismo/knx-go/knx/cemi#LDataInd), than you have the same behavior.

telemac, 2017-07-03:
Ok, i did it this way :

package main

import (
	"log"
	"os"
	"time"

	"github.com/vapourismo/knx-go/knx"
	"github.com/vapourismo/knx-go/knx/cemi"
	"github.com/vapourismo/knx-go/knx/knxnet"
	"github.com/vapourismo/knx-go/knx/util"
)

var defaultGroupLData = cemi.LData{
	Control1: cemi.Control1NoRepeat | cemi.Control1NoSysBroadcast | cemi.Control1WantAck | cemi.Control1Prio(cemi.PrioLow),
	Control2: cemi.Control2GrpAddr | cemi.Control2Hops(6),
}

// buildGroupOutbound constructs the L_Data core frame for group communication.
func buildGroupOutbound(src cemi.IndividualAddr, dest cemi.GroupAddr, data []byte) cemi.LData {
	ldata := defaultGroupLData
	ldata.Data = &cemi.AppData{Command: cemi.GroupValueWrite, Data: data}
	ldata.Source = src
	ldata.Destination = uint16(dest)

	if len(data) <= 15 {
		ldata.Control1 |= cemi.Control1StdFrame
	}

	return ldata
}

func Send(tunnel *knx.Tunnel, src cemi.IndividualAddr, dest cemi.GroupAddr, data []byte) error {
	return tunnel.Send(&cemi.LDataReq{LData: buildGroupOutbound(src, dest, data)})
}

func main() {
	util.Logger = log.New(os.Stdout, "", log.LstdFlags)

	tunnel, err := knx.NewTunnel("192.168.1.35:3671", knxnet.TunnelLayerData, knx.DefaultTunnelConfig)
	if err != nil {
		panic(err)
	}

	defer tunnel.Close()

	var value uint8
	for i := 0; i < 10; i++ {
		value = 1 - value
		err = Send(tunnel, cemi.NewIndividualAddr3(1, 2, 3), cemi.NewGroupAddr3(1, 1, 3), []byte{value})
		time.Sleep(105 * time.Millisecond)
	}

	for msg := range tunnel.Inbound() {

		if ind, ok := msg.(*cemi.LDataInd); ok {
			if app, ok := ind.Data.(*cemi.AppData); ok {
				gc := knx.GroupComm{
					Source:      ind.Source,
					Destination: cemi.GroupAddr(ind.Destination),
					Data:        app.Data,
				}
				var command string
				switch app.Command {
				case cemi.GroupValueResponse:
					command = "RESPONSE"
					break
				case cemi.GroupValueRead:
					command = "READ"
					break
				case cemi.GroupValueWrite:
					command = "WRITE"
					break
				default:
					command = "OTHER"
					break
				}
				log.Printf("command = %s, source = %s, destination = %s; ", command, gc.Source, gc.Destination)
			} else {
				util.Logger.Printf("Received L_Data.ind frame does not contain application data")
			}
		} else {
			util.Logger.Printf("Received frame is not a L_Data.ind frame %T %+v", msg, msg)
		}

	}

}

i had to copy/paste defaultGroupLData and buildGroupOutbound, do you think there is an interest to export them ?
Thank you very much for your awesome work and help
Alexandre

closed 2017-07-03 by telemac

vapourismo, 2017-07-03:
What are you trying to accomplish? As far as I can tell, you've only added the ability to logging.

telemac, 2017-07-03:
I only added logging to avoid pointless code, and i come form c++, probably the origin of the useless break.

I'd like to be able to determine if a group address is readable or not without having the ETS project, and trace the activity on the bus with a bit more detail. To diagnose a problem, say a readable address that don't answer, it is a good thing to see the read on the bus.

Actually it does the work.

Instead of exporting the constant, exporting a function returning the constant could be a workaround may be.

vapourismo, 2017-07-03:
Even if the device's communication object refs have been programmed with the Read-flag enabled, devices don't have to respond to read requests. So it is very unreliable.

In general you only want to listen to group events that contain data. Virtually every device broadcasts changes to group objects.

Anyway, I will rework the `GroupComm` a bit to accomodate for scenarios where it is required to react to read requests.

vapourismo, 2017-07-04:
Have a look at [knx.GroupEvent](https://godoc.org/github.com/vapourismo/knx-go/knx#GroupEvent).
I've reworked it, so it should now accurately resemble group events.