vapourismo/knx-go #3: make a cemi.GroupAddr form an a/b or a/b/c string

telemac, 2017-07-03:
I needed this function, may be it can be useful to others

```
// NewGroupAddrFromString returns a cemi.GroupAddr from a group address string a/b or a/b/c
func NewGroupAddrFromString(groupAddress string) (cemi.GroupAddr, error) {
	parts := strings.Split(groupAddress, "/")
	if len(parts) < 2 || len(parts) > 3 {
		return 0, fmt.Errorf("%s is not a valid group address", groupAddress)
	}
	n := make([]int, 0, len(parts))
	for _, part := range parts {
		p, err := strconv.Atoi(part)
		if err != nil {
			return 0, fmt.Errorf("%s is not a valid group address", groupAddress)
		}
		n = append(n, p)
	}
	if len(n) == 2 {
		return cemi.NewGroupAddr2(uint8(n[0]), uint8(n[1])), nil
	}
	return cemi.NewGroupAddr3(uint8(n[0]), uint8(n[1]), uint8(n[2])), nil
}

```

vapourismo, 2017-07-03:
You can use `fmt.Sscanf`.

closed 2017-07-03 by vapourismo

telemac, 2017-07-04:
right.
```
// NewGroupAddrFromString returns a cemi.GroupAddr from a group address string a/b or a/b/c
func NewGroupAddrFromString(groupAddress string) (cemi.GroupAddr, error) {
	var a, b, c int
	n, err := fmt.Sscanf(groupAddress, "%d/%d/%d", &a, &b, &c)
	if err == nil && n == 3 {
		return cemi.NewGroupAddr3(uint8(a), uint8(b), uint8(c)), nil
	}
	n, err = fmt.Sscanf(groupAddress, "%d/%d", &a, &b)
	log.Printf("Sscanf %d %v %d %d %d", n, err, a, b, c)
	if err == nil && n == 2 {
		return cemi.NewGroupAddr2(uint8(a), uint8(b)), nil
	}
	return 0, fmt.Errorf("%s is not a valid group address", groupAddress)
}
```

vapourismo, 2017-07-04:
I don't think you have to call `Sscanf` twice. Just check whether `n` is 1, 2 or 3.

vapourismo, 2017-07-04:
Now supported in cc180e2e3f7b6841b756cad06b4f7e6546c542e6.

Documentation:
 - [cemi.NewGroupAddrString](https://godoc.org/github.com/vapourismo/knx-go/knx/cemi#NewGroupAddrString)
 - [cemi.NewIndividualAddrString](https://godoc.org/github.com/vapourismo/knx-go/knx/cemi#NewIndividualAddrString)