vapourismo/knx-go #15: How to retry if not connected

ferllings, 2019-03-05:
Hello,
I have a N148/21 interface, that only supports 1 tunnelling connection.

When the connection is already taken, the interface returns a "NO_MORE_CONNECTIONS" error (code 0x24)
With the default configuration it times out after a 10 secs

Is it possible to detect it and retry after 1 min?

Thanks

vapourismo, 2019-03-05:
That's already the case. The tunnelling connection understands the 0x24/0x25 error codes, and will continue retrying.

If you want it to retry longer than 10s, you'll have to override the default response timeout. Look at the TunnelConfig you pass in.

ferllings, 2019-03-05:
Ah, ok, I see.

Thanks a lot!

closed 2019-03-05 by ferllings

vapourismo, 2019-03-05:
Now that I think about it, maybe `ResponseTimeout` is not very intuitive as it is used for all response timeouts not just the connection establishment.

I'll add a `ConnectionTimeout` if I have time for it.