vapourismo/knx-go #16: How to achieve resilience

ferllings, 2019-03-06:
Hello,

I have another question, I hope this is the right place to ask.

As I previously mentioned on another ticket, I have a N148/21 with only one tunnel connection.

If the gateway connection is not available (it can happen if the previous connection is not properly disconnected) the gateway will be unavailable for 1 min.
So I'm using `ResendInterval:    60 * time.Second,`

But because the gateway and the go app are not on the same network, the connection might be lost for several minutes.
So I'm using `ResponseTimeout:   1 * time.Hour` to any (temporary) network issue

But does the lib try to reconnect automatically if the connection is lost?

Thanks,

vapourismo, 2019-03-07:
A resend interval and response timeout that large are not recommendable.
Imagine a frame is lost, that means it will be resend only a minute later.

And if the heartbeat fails during that time, the connection will be terminated.

From the protocols perspective, it is almost impossible to tell if the device is just busy, dead or there is a network issue.

The library is fairly low-level. It doesn't do reconnects.

I suggest building something on top which does rebuild the tunnel client when it fails.

ferllings, 2019-03-07:
Thanks,

so I should use a shorter resendInterval (500 mili), and something like 2mins for the responseTimeout.

Then handle the disconnect/unavailable gateway on my app.

vapourismo, 2019-03-07:
I image it would be a channel-like abstraction on top of the `TunnelClient` that on repeated failure to send or on the inbound channel being closed (which indicates a closed connection) simply discards the `TunnelClient` and instantiates a new one. Possibly needs to make the `errResponseTimeout` error public so you can compare against it. That would be a simple solution.

I've had a bit of time to think about it and I think this can actually be integrated into the `TunnelClient` directly. However it wouldn't be too easy and I have neither the time nor the motivation nor the means to test it. In any case I am happy to accept PRs though :smile:

closed 2019-03-07 by vapourismo

vapourismo, 2019-03-07:
Actually, I think the `TunnelClient` already tries reconnect: https://github.com/vapourismo/knx-go/blob/master/knx/tunnel.go#L508

However, it only retries once. So what is needed is to extend the TunnelConfig to support a reconnect attempt counter.