vapourismo/knx-go #17: Questions towards AppData.pack

stsaid, 2019-07-07:
tpdu.go contains the following lines in `func (app *AppData) Pack(buffer []byte)`:
```
buffer[2] &= 63
buffer[2] |= byte(app.Command&3) << 6
```
What do they do?

I wanted to implement DPT 16.000/16.001.
Because of `copy(buffer[2:], app.Data)`, `buffer[2:16]` contain a 14 Byte String.
The mentioned lines overwrite the first byte in `buffer[2]`.

I guess, I lack knowledge on the basics of assembling KNX datagrams.

Thanks for your reply.

vapourismo, 2019-07-08:
The contents you pass in `Data` are ill-formated. When you send more than 6bits in the `Data` field, you must keep the first byte (`app.Data[0]`) untouched. The reason for this is: KNX protocol is not very programmer friendly and Go can't easily enforce this staticly.

Have a peek at this: https://openremote.github.io/archive-dotorg/forums/CEMI%20APDU%20Field.html
Specifically this:
```
+--------+--------+--------+--------+--------+
| TPCI + | APCI + |  Data  |  Data  |  Data  |
|  APCI  |  Data  |        |        |        |
+--------+--------+--------+--------+--------+
  byte 1   byte 2   byte 3     ...    byte 16
```

"byte 2" in this example corresponds to `buffer[2]` in the code. The 2 most significant bits of it store parts of the APCI (e.g. group write, read, response, etc.). Hence "byte 2" only has 6 bits available for the user. The code above takes care of injecting the APCI.

closed 2019-07-08 by vapourismo

stsaid, 2019-07-09:
Thanks! I will check that out.