vapourismo/knx-go #20: Communicating directly with a KNX serial gateway

gabrielruiu, 2019-08-05:
This library currently supports only KNXnet/IP. I am interested in extending the support for my application, which should also be able to directly communicate to serial KNX gateway/devices (which are mounted on a system as a `/dev/tty*` device).

My question is: How would I go about doing that?

I've noticed in the code that the `Socket` interface might be the solution. To just replace the current `TunnelSocket` implementation with one that writes directly to a local serial device.

From you experience with KNX, would this work? Are the `[]byte` payloads reusable in both cases?

vapourismo, 2019-08-05:
From what I understand, devices that are connected via USB or serially are vastly different from KNXnet/IP devices.

It would be inappropriate to reuse the Socket interface given that the communication doesn't run via sockets.

I would imagine you could reuse the `knx/dpt` for all datapoint type related things. It gets tricky with 
`knx/cemi` because devices might use EMI1 or EMI2 frames which are not the same.

The very first step should be an analysis of how the frames to those devices actually look like.

gabrielruiu, 2019-08-05:
Thank you for the quick reply. Will investigate further.

closed 2019-08-05 by gabrielruiu

ferllings, 2022-03-22:
@gabrielruiu Did you find a way to use serial with the library?
I'm looking for doing the same.