vapourismo/knx-go #23: DPT 5.001 as float32

ferllings, 2019-09-06:
Hello,

I believe in https://github.com/vapourismo/knx-go/blob/6d1e5b4779371b381cf8468f98fafd7340b45809/knx/dpt/types.go#L141

should be a 8-bit unsigned value, as the DPT value is 0...100 percentage.
```
type DPT_5001 uint8
```

vapourismo, 2019-09-06:
DPT 5.001 has a resolution of about 0.4%. A `uint8` can't represent that.

closed 2019-09-06 by vapourismo

ferllings, 2019-09-06:
Unless I misunderstand the specs says

"5.yyy = 8-bit unsigned value, like dim value (0..100%), blinds position (0..100%)"

https://support.knx.org/hc/en-us/articles/115001133744-Datapoint-Type

The payload is only 1byte so not enough for float value I believe.

ferllings, 2019-09-07:
I'm happy to provide a pull request if you agree.

vapourismo, 2019-09-07:
That is the wire representation.
The type DPT_5001 is an abstraction on top of it. One should have noticed this because its range is 0..100 including decimals and not the wire 8-bit range of 0..255 without decimals.

ferllings, 2019-09-07:
Ah, I see it now...

I was trying to find the reason for the difference between the value I send to my shutter (30%)
And the returned value (29.80%)

And you're right:

Using Wireshark, the returned value is "4c" (hexadecimal)
Which gives "76" (0...255 range) so  "29.8039" (0...100 range)

So ETS (and others) are applying a round on the value.

Thanks for explaining.