vapourismo/knx-go #31: KNX virtual crash when Write Command send

maarthala, 2020-08-27:
Hi,
I am doing POC with this library for one of the upcoming project.
I have a KNX virtual running ( https://www.knx.org/knx-en/for-professionals/get-started/knx-virtual/index.php ) in windows box.
When i try to use this package and control a switch group, KNX virtual is crashing. I dont see any issue when i try with Python library . ( https://xknx.io/ )

Can it be a problem  with Virutal device which unable to intercept go request ?  Will there be similar impact when we deal with real devices ?. For business reasons, we have to use go language.  Any comment or help on this ?

Go Code:
data := dpt.DPT_1001(true).Pack()
	err = client.Send(knx.GroupEvent{
		Command:     knx.GroupWrite,
		Destination: cemi.NewGroupAddr3(1, 1, 1),
		Data:        data,
	})

Python Code:
switch = Switch(xknx,name='ch1',group_address='1/1/1')
await switch.set_on()
await xknx.stop()

Thanks,

vapourismo, 2020-08-27:
You need to provide a little bit more information than that.

Do you use a router or tunnel client?
Can you provide a more detailed example? 
Have you brought this up with the ‘knx-virtual’ project? 
Does ‘knx-virtual’ log any information before crashing?

I have not experienced any KNX device (or any software for that matter) crashing when used with this library.

maarthala, 2020-08-27:
Thanks in advance. @vapourismo 
Please let me know if you need any further information. I really appreciate your help.

Extra information:
I have setup HomeAssistant (https://www.home-assistant.io/) in IP 192.168.1.84 (Mac) and configured KNX Virtual. I am able to control the Switch without any issues. 

I using Tunnel client
client, err := knx.NewGroupTunnel("192.168.1.21:3671", knx.DefaultTunnelConfig)

Can you provide a more detailed example?
Using KNX ETS i have setup a lamp and button with group "1/1/1". I am trying to control the switch using this package.
KNX Virtual installed in a separate windows box with IP 192.168.1.21.
Go/python code is running in mac with IP 192.168.1.84. They both are in the same network.

Have you brought this up with the ‘knx-virtual’ project?
Not yet. As this was working fine when used with Python and HomeAssistant.

Does ‘knx-virtual’ log any information before crashing?
No info before crashing but i found this in event log of windows.
Log Name:      Application
Source:        .NET Runtime
Date:          27/08/2020 12:05:49
Event ID:      1026
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-8L3MTNC
Description:
Application: KNX Virtual.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Net.Sockets.SocketException
   at System.Net.Sockets.Socket.Send(Byte[], Int32, Int32, System.Net.Sockets.SocketFlags)
   at System.Net.Sockets.UdpClient.Send(Byte[], Int32, System.Net.IPEndPoint)
   at KnxVirtualForm.MainForm.CoreService(Byte[])
   at KnxVirtualForm.MainForm.ListenUdpClient(System.IAsyncResult)
   at System.Net.LazyAsyncResult.Complete(IntPtr)
   at System.Net.ContextAwareResult.CompleteCallback(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Net.ContextAwareResult.Complete(IntPtr)
   at System.Net.LazyAsyncResult.ProtectedInvokeCallback(System.Object, IntPtr)
   at System.Net.Sockets.BaseOverlappedAsyncResult.CompletionPortCallback(UInt32, UInt32, System.Threading.NativeOverlapped*)
   at System.Threading._IOCompletionCallback.PerformIOCompletionCallback(UInt32, UInt32, System.Threading.NativeOverlapped*)

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name=".NET Runtime" />
    <EventID Qualifiers="0">1026</EventID>
    <Level>2</Level>
    <Task>0</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2020-08-27T09:05:49.900112100Z" />
    <EventRecordID>35913</EventRecordID>
    <Channel>Application</Channel>
    <Computer>DESKTOP-8L3MTNC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Application: KNX Virtual.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Net.Sockets.SocketException
   at System.Net.Sockets.Socket.Send(Byte[], Int32, Int32, System.Net.Sockets.SocketFlags)
   at System.Net.Sockets.UdpClient.Send(Byte[], Int32, System.Net.IPEndPoint)
   at KnxVirtualForm.MainForm.CoreService(Byte[])
   at KnxVirtualForm.MainForm.ListenUdpClient(System.IAsyncResult)
   at System.Net.LazyAsyncResult.Complete(IntPtr)
   at System.Net.ContextAwareResult.CompleteCallback(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Net.ContextAwareResult.Complete(IntPtr)
   at System.Net.LazyAsyncResult.ProtectedInvokeCallback(System.Object, IntPtr)
   at System.Net.Sockets.BaseOverlappedAsyncResult.CompletionPortCallback(UInt32, UInt32, System.Threading.NativeOverlapped*)
   at System.Threading._IOCompletionCallback.PerformIOCompletionCallback(UInt32, UInt32, System.Threading.NativeOverlapped*)

</Data>
  </EventData>
</Event>

maarthala, 2020-08-27:
Here i am attaching Go code and ETS5 sample knx project as zip file.

[knx.zip](https://github.com/vapourismo/knx-go/files/5135935/knx.zip)

vapourismo, 2020-08-29:
This seems like a KNX Virtual issue so far.

Can you provide network captures so we can look at the differences between `xknx` and this library?
Ideally something Wireshark-compatible.

closed 2020-09-24 by vapourismo