vapourismo/knx-go #33: working with DPT 7 (2-Octet Unsigned Value) .007 (DPT_TimePeriodHrs)

LukasGrebe, 2020-11-22:
Hi thanks you so much for this repository! I'm new to working with KNX on such a close-to-wire level.
I'm wondering how to work with  DPT 7.007 values: they arn't specifically mentioned in the code. I also saw issue #7 mentioning https://godoc.org/github.com/vapourismo/knx-go/knx#GroupEvent
yet i'm unsure how to proceed.

vapourismo, 2020-11-22:
None of the DPT 7.* types have been implemented, yet. You can easily implement them yourself and add to [knx/dpt](https://pkg.go.dev/github.com/vapourismo/knx-go@v0.0.0-20200924173532-6ed2bf1e51e6/knx/dpt) if you want. All types should support the listed `Pack`, `Unpack`, `Unit` and `String` methods. The repo README.md has two examples how they are supposed to be used.

With all DPT types, you need figure out its representation. You can usually find this [here](https://www.knx.org/wAssets/docs/downloads/Certification/Interworking-Datapoint-types/03_07_02-Datapoint-Types-v02.01.02-AS.pdf).

As for DPT 7.*, they are represented as an unsigned 16-bit integer. This means the `Data` field in the `GroupEvent` structure that you linked will consist of 3 bytes, the leading empty byte (because the payload is bigger than 6 bits) and the trailing 2 bytes which represent the 16 bits of payload from the unsigned integer.

closed 2020-11-22 by vapourismo

alexander-zimmermann, 2021-08-15:
Implemented in pull request #35