vapourismo/knx-go #34: DPT 251.600

MitchSUEW, 2021-04-08:
Hi,
will there be a support for DPT 251.600, 6 bytes. It's a RGBW LED connected through a DALI KNX adapter from Hager (TXA664D).
Thanks,
Mitch

vapourismo, 2021-04-08:
I currently don't have any plans to implement it. 
Pull requests are welcome, though. Documentation is [here](https://www.knx.org/wAssets/docs/downloads/Certification/Interworking-Datapoint-types/03_07_02-Datapoint-Types-v02.01.02-AS.pdf) (section 6.18 covers DPT 251.600).

vapourismo, 2021-04-08:
If you want to give this a shot yourself, there are plenty of PRs that added new DPTs. I think they act as a good base guidance. Otherwise feel free to reach to me directly, if you have more questions.

closed 2021-04-08 by vapourismo

MitchSUEW, 2021-04-08:
Thx. I don't think I'm capable to do so.(at least for the time being). But I'm confident there's someone out there...... :-)

korpa, 2021-11-28:
@MitchSUEW I added DPT 251.600 in #4.

As soon as @vapourismo accepted the PR you should be able to use it.

MitchSUEW, 2021-11-29:
Wow, that’s awesome. Looking forward to it.

korpa, 2021-11-29:
@MitchSUEW: @vapourismo accepted the pull request yesterday so DPT 251.600 support is in the master. A `go get github.com/vapourismo/knx-go` in your project should download the latest version.

An example how it works can be found in #41