vapourismo/knx-go #39: MessageCode differs on GOARM 5

jurjevic, 2021-08-29:
The received message on the inbound seems to be different on different architecture. So on my Mac (AMD64) the casting to LDataInd works for all incoming messages. On my BeagleBoneBlack I use the ARM 5 architecture, which receives some messages of type LDataCon.

**Build options:**
`GOOS=linux GOARCH=arm GOARM=5`

**Function:**
`func serveGroupInbound(inbound <-chan cemi.Message, outbound chan<- GroupEvent)`

**Code:**
`if ind, ok := msg.(*cemi.LDataInd); ok`

**Workaround:** Adding additional check like...
`if con, ok := msg.(*cemi.LDataCon); ok`

Is this an appropriate solution?

vapourismo, 2021-08-29:
I am not quite sure what you're describing, unless I am misunderstanding, is exactly possible. I don't think received message codes differ, which would make the casts fail, because of the underlying architecture. I can't exclude something weird going on, but so far I think it is unlikely.

Is it possible that you're arm5 machine sends some `LData.req` frames that get responded to with `LData.con` frames?

jurjevic, 2021-08-30:
Unfortunately this seems to be the case. On my dev machine (Mac) and BBB the inbound contains different types for the same received knx message. So the incoming LData messages, but must always be a LDataInd? In which case a LDataCon would be Ok? (just trying to understand the difference)

vapourismo, 2021-08-30:
Can you verify the network traffic is actually identical using a network package capture to ensure the problem is actually with this package?

If you could provide a minimal complete code example of what you're doing exactly on each machine would help a lot.

jurjevic, 2021-09-01:
Ok. So I traced it down using diagnose on my knx bus.

The good...
- The message code on the bus was always LDataInd
- The arm and mac code work the same (luckily :-)

The bad...
When I write to a group address (which works), then the written knx message on the same machine is received in `inbound <-chan cemi.Message` as **LDataCon** - on a different machine (but same code) it is received as LDataInd. I would assume that this is somehow wrong. 

What I expected...
The written message should also be of type **LDataInd** in the inbound. Or am I missing something?
E.g. I've one app with "virtual knx components". One virtual knx component writes a state update to the bus and another virtual knx component is notified by listening to the inbound channel (in the same app).

Btw... the reason for my bad assumption was, that my app running as a single instance has this issue. But when I develop on my Mac, I run a second instance of the app... this hides the issue, because  one instance writes and the other instance gets notified and vice versa.

So thats was the reason for my bad assumption.

vapourismo, 2021-09-01:
Without any code or a detailed description of what your setup is exactly this is quite hard to debug as I have to guess what you're doing.

I am guessing both your machines attach to a KNX network via a gateway? Are you using the `GroupTunnel`?
If so, it is correct behaviour that the instance that sends the LData *req*uest also gets the LData *con*firmation. In contrast to the requester, the other instance which did not fire a request is only an observing party and therefore gets the LData *ind*ication.

jurjevic, 2021-09-02:
Yes, you are right. When writing to the bus the correct behavior is to receive to get 'LDataCon' message (I use GroupTunnel).
I'm new to the EIB protocol, but I'm trying digging down. So the only left question is, why don't you pass the 'LDataCon' messages coming from the inbound also to the outbound channel? It is a valid group message with just a different message code.
```go
// serveGroupInbound serves a group communication.
func serveGroupInbound(inbound <-chan cemi.Message, outbound chan<- GroupEvent) {
	util.Log(inbound, "Started worker")
	defer util.Log(inbound, "Worker exited")

	for msg := range inbound {
		if ind, ok := msg.(*cemi.LDataInd); ok {
			...
			} else {
				util.Log(inbound, "Received L_Data.ind frame does not contain application data")
			}
		} else {
			util.Log(inbound, "Received frame is not a L_Data.ind frame")
		}
	}

	close(outbound)
}
```

vapourismo, 2021-09-03:
I don't quite remember why that is. However, I am relatively sure it is to avoid feedback loops. That is only indications are meant to be received from the gateway and relayed to the user of this package.

If one were to point me to a section from the KNX standard that states that LData.con frames are also meant to be treated as incoming data, I would happily facilitate a change in this package to conform with that.

jurjevic, 2021-09-04:
Well, I've not found any information on that in the spec that I read. I've checked some other implementations and tested some of my devices.

The diagnose tool 'group monitor' in ETS is also able to write data. The written messages are echoed correctly with an L_Data Con messaged id. So the written message is also received. This may just be for debugging reason, so it's hard to state this as a reference.

But MDT and Merten devices are listening to ga which are also sent by the same device. E.g. a switch device with multiple channels, can be configured that an emitted status message of one channel can be set as an input message for another channel. While this is not a "clear" evidence, it is an indication.

Another code that I checked was the use examples from knxd. I've combined the groupsocketwrite.c and groupsocketlisten.c code to test it. The provided examples do not filter message ids. It's up to the 'consumer' and I've received also the sent group messages (I know, it's a very low level example implementation).

A 'listen on write for same ga' should not produce a loop automatically. Only if someone manually implements it, but in this case it's up to the api consumer to take care.  By the way, 'looping' is also possible by the ETS... so the user is responsible for that.

What do you think?

vapourismo, 2021-09-06:
I am not convinced. None of this is an argument for also relaying data in LData.con frames.

ETS' group monitor operates at a lower level, hence you'll see all types of CEMI frames. This is also true for your knxd example.
The [Tunnel type](https://pkg.go.dev/github.com/vapourismo/knx-go@v0.0.0-20210818194629-bb58218a6e2e/knx#Tunnel) lets you do lower level communication, if you want to do that.

On-line devices communicate differently. Whether they can read their own writes is fairly unrelated to this discussion.

As for loop back detection, it is not always feasible to implement this as a user as the user has no information on whether the frame was an indication or a confirmation. 

I am going to spend some time trying to get a hold of the KNX Standard and read it in order to find out what is the correct behaviour.
In the meanwhile, I think a good workaround is to make this type of behaviour configurable in the [TunnelConfig](https://pkg.go.dev/github.com/vapourismo/knx-go@v0.0.0-20210818194629-bb58218a6e2e/knx#TunnelConfig). This way one can toggle receiving data from confirmation frames.

jurjevic, 2021-09-07:
This sounds like a good solution. The TunnelConfig is the right place for this.

closed 2021-09-07 by jurjevic