vapourismo/knx-go #46: Impossible to read memory address with Authentication

sebglon, 2022-03-28:
The authReq 0x3D1 is not supported.
Can you help me to implement it?
This seems the case for all extended AL Codes

Wireshark description is available here: https://github.com/wireshark/wireshark/blob/499a33c7092cb20d18e6194e09bb6f704e716076/epan/dissectors/packet-cemi.c#L430

sebglon, 2022-03-29:
This extended APCI is used in secencial call when we want to read or write a data in a memory block at level for example.
THis is the BCU password on ETS 5.
I have trying to add the APCI on your link but I have many other error due to the frame size.
And I don"t know to generate a secencial call

example: 
Send authReq
receive an ACK
caller send an AuthReq
I emit an ACK
caller send an AuthRest with the granted access level
I emit an ACK
A t this step I can read all memory address if level is 0

```
550	2.499290	192.168.1.29	192.168.1.2	KNXnet/IP	68	Tunnel ConnectReq @0.0.0.0:0 @0.0.0.0:0
560	2.501364	192.168.1.2	192.168.1.29	KNXnet/IP	62	Tunnel ConnectResp #C3 @0.0.0.0:0, 1.1.100
604	2.521418	192.168.1.29	192.168.1.2	KNXnet/IP	66	Config ConnectReq @0.0.0.0:0 @0.0.0.0:0
605	2.523237	192.168.1.2	192.168.1.29	KNXnet/IP	60	Config ConnectResp #C4 @0.0.0.0:0
606	2.526156	192.168.1.29	192.168.1.2	KNXnet/IP	59	ConfigReq #C4:0 M_PropRead.req OT=8 P=51
607	2.528573	192.168.1.2	192.168.1.29	KNXnet/IP	60	ConfigAck #C4:0 OK
608	2.529287	192.168.1.2	192.168.1.29	KNXnet/IP	61	ConfigReq #C4:0 M_PropRead.con OT=8 P=51 $0002
610	2.529598	192.168.1.29	192.168.1.2	KNXnet/IP	52	ConfigAck #C4:0 OK
611	2.531724	192.168.1.29	192.168.1.2	KNXnet/IP	59	ConfigReq #C4:1 M_PropRead.req OT=0 P=56
612	2.533702	192.168.1.2	192.168.1.29	KNXnet/IP	60	ConfigAck #C4:1 OK
613	2.534419	192.168.1.2	192.168.1.29	KNXnet/IP	61	ConfigReq #C4:1 M_PropRead.con OT=0 P=56 $0037
614	2.534512	192.168.1.29	192.168.1.2	KNXnet/IP	52	ConfigAck #C4:1 OK
615	2.536084	192.168.1.29	192.168.1.2	KNXnet/IP	58	DisconnectReq #C4 @0.0.0.0:0
616	2.537776	192.168.1.2	192.168.1.29	KNXnet/IP	60	DisconnectResp #C4 OK
674	2.577121	192.168.1.29	192.168.1.2	KNXnet/IP	62	TunnelReq #C3:0 L_Data.req 0.0.0->1.1.18 Connect
675	2.578995	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:0 OK
716	2.616445	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:0 L_Data.con 1.1.100->1.1.18 Connect
717	2.616864	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:0 OK
722	2.619728	192.168.1.29	192.168.1.2	KNXnet/IP	68	TunnelReq #C3:1 L_Data.req 0.0.0->1.1.18 AuthReq $FFFAAEE5
725	2.621581	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:1 OK
727	2.647529	192.168.1.2	192.168.1.29	KNXnet/IP	68	TunnelReq #C3:1 L_Data.con 1.1.100->1.1.18 AuthReq $FFFAAEE5
728	2.647722	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:1 OK
729	2.678025	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:2 L_Data.ind 1.1.18->1.1.100 ACK
730	2.678168	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:2 OK
732	2.710536	192.168.1.2	192.168.1.29	KNXnet/IP	64	TunnelReq #C3:3 L_Data.ind 1.1.18->1.1.100 AuthResp L=15
733	2.710741	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:3 OK
734	2.711628	192.168.1.29	192.168.1.2	KNXnet/IP	62	TunnelReq #C3:2 L_Data.req 0.0.0->1.1.18 ACK
735	2.713680	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:2 OK
736	2.730391	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:4 L_Data.con 1.1.100->1.1.18 ACK
737	2.730568	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:4 OK
774	2.789750	192.168.1.29	192.168.1.2	KNXnet/IP	67	TunnelReq #C3:3 L_Data.req 0.0.0->1.1.18 PropValueRead OX=0 P=56
780	2.792470	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:3 OK
791	2.817493	192.168.1.2	192.168.1.29	KNXnet/IP	67	TunnelReq #C3:5 L_Data.con 1.1.100->1.1.18 PropValueRead OX=0 P=56
792	2.817660	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:5 OK
800	2.849145	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:6 L_Data.ind 1.1.18->1.1.100 ACK
801	2.849344	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:6 OK
803	2.888798	192.168.1.2	192.168.1.29	KNXnet/IP	67	TunnelReq #C3:7 L_Data.ind 1.1.18->1.1.100 PropValueResp OX=0 P=56 N=0
804	2.889127	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:7 OK
805	2.889603	192.168.1.29	192.168.1.2	KNXnet/IP	62	TunnelReq #C3:4 L_Data.req 0.0.0->1.1.18 ACK
806	2.892069	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:4 OK
807	2.908603	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:8 L_Data.con 1.1.100->1.1.18 ACK
808	2.908913	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:8 OK
809	2.912783	192.168.1.29	192.168.1.2	KNXnet/IP	70	TunnelReq #C3:5 L_Data.req 0.0.0->1.1.18 PropValueWrite OX=0 P=8 $0FFFFC
810	2.914618	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:5 OK
814	2.944471	192.168.1.2	192.168.1.29	KNXnet/IP	70	TunnelReq #C3:9 L_Data.con 1.1.100->1.1.18 PropValueWrite OX=0 P=8 $0FFFFC
815	2.944718	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:9 OK
816	2.972429	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:10 L_Data.ind 1.1.18->1.1.100 ACK
817	2.972637	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:10 OK
819	3.018411	192.168.1.2	192.168.1.29	KNXnet/IP	67	TunnelReq #C3:11 L_Data.ind 1.1.18->1.1.100 PropValueResp OX=0 P=8 N=0
820	3.018848	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:11 OK
821	3.019464	192.168.1.29	192.168.1.2	KNXnet/IP	62	TunnelReq #C3:6 L_Data.req 0.0.0->1.1.18 ACK
822	3.021935	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:6 OK
823	3.033940	192.168.1.29	192.168.1.2	KNXnet/IP	62	TunnelReq #C3:7 L_Data.req 0.0.0->1.1.18 Disconnect
824	3.036139	192.168.1.2	192.168.1.29	KNXnet/IP	60	TunnelAck #C3:7 OK
825	3.039263	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:12 L_Data.con 1.1.100->1.1.18 ACK
826	3.039706	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:12 OK
827	3.057381	192.168.1.2	192.168.1.29	KNXnet/IP	62	TunnelReq #C3:13 L_Data.con 1.1.100->1.1.18 Disconnect
828	3.057797	192.168.1.29	192.168.1.2	KNXnet/IP	52	TunnelAck #C3:13 OK
829	3.065302	192.168.1.29	192.168.1.2	KNXnet/IP	58	DisconnectReq #C3 @0.0.0.0:0
830	3.067229	192.168.1.2	192.168.1.29	KNXnet/IP	60	DisconnectResp #C3 OK
```

vapourismo, 2022-03-29:
I don't understand that last message. If you have an error, do you mind posting the error?

What do you mean by "secencial call"?

The log you attached seems to confirm that the functionality around LData frames is already sufficient to achieve what you want.

closed 2022-04-12 by vapourismo