vapourismo/knx-go #55: Implement DESCRIPTION_REQUEST

theobarrague, 2022-07-27:
I want to send a DESCRIPTION_REQUEST over IP Gateway.
I can't find an implementation, did you implemented it ? If no, I can help

mobilarte, 2022-07-27:
DESCRIPTION_REQUEST or DIAGNOSTIC_REQUEST are not implemented here. I have it on my todo list, but I have currently a pull request that needs to be fixed (Multicast behavior on Windows).

I have an experimental copy of this implementation with more DPTs, where I also rewrote the parsing of DIBs including DESCRIPTION_REQUESTs. It is available at [knx-exp](https://github.com/mobilarte/knx-exp). Feel free to test it.

vapourismo, 2022-07-28:
@mobilarte Want me to assign you this issue for tracking?

mobilarte, 2022-07-29:
This will probably not be necessary. I have just added a PR for DESCRIPTION_REQUEST with a few cosmetic fixes. Thank you for reviewing.

vapourismo, 2022-08-08:
This has been addressed in #56 thanks to @mobilarte.

closed 2022-08-08 by vapourismo