vapourismo/knx-go #65: Query value-status of a group address

rltsv, 2023-01-09:
Hi!

Is it possible to implement a status query of all group addresses connected to the gateway? Or, for example, without waiting for the status change, just request the current value of the group address?

At the moment, we get the status value only by changing it.

mobilarte, 2023-01-09:
I assume that you have the list of configured group addresses. If you send a `GroupRead` event to a group address, you should get a reply (`GroupResponse`) from one of the devices that uses the address.
See also [GroupValueRead: from the bus](https://support.knx.org/hc/en-us/articles/115003188669-GroupValueRead-from-the-bus) for more details about which device will reply.

vapourismo, 2023-01-09:
As @mobilarte has pointed out, the network doesn't freely tell you which addresses are allocated, unfortunately. That means you would need to know every used address upfront.

With enough time you could brute force query *every* possible address upfront. Not a good idea IMHO though.

If you have all the addresses in question, you can issue `GroupValueRead`s for each of them. That would give you an asynchronous stream of values for each of them.

closed 2023-01-16 by vapourismo