vapourismo/knx-go #67: support for knx secure tunnel?

pescobar, 2023-01-23:
Hi,

I am using a [weinzierl device](https://weinzierl.de/en/products/knx-ip-interface-732-secure/) with security enabled. I can query it using the xknx python library and an example similar [to this one](https://github.com/XKNX/xknx/blob/main/examples/example_secure.py). This is my working test:

```
import asyncio
import logging

from xknx import XKNX
from xknx.io import ConnectionConfig, ConnectionType, SecureConfig
from xknx.tools import group_value_write
from xknx.devices import Light

logging.basicConfig(level=logging.INFO)
logging.getLogger("xknx.log").level = logging.DEBUG
logging.getLogger("xknx.knx").level = logging.DEBUG

async def main() -> None:
    """Test connection with IP secure tunnelling."""

    connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING_TCP_SECURE,
        gateway_ip="192.168.1.44",
        individual_address="1.0.11",
        secure_config=SecureConfig(
            user_id=1,
            device_authentication_password="*******",
            user_password="***********",
        ),
    )
    xknx = XKNX(connection_config=connection_config)

    await xknx.start()
    print("Tunnel connected")

    # Connect to KNX/IP bus, switch on light, wait 2 seconds and switch of off again
    light = Light(xknx, name="TestLight", group_address_switch="1/1/5")
    await light.set_on()
    await asyncio.sleep(2)
    await light.set_off()

    await xknx.stop()

asyncio.run(main())
```

I would like to query the same device using telegraf to push some metrics to influxdb but I couldn't find any config example to connect to a secure tunnel. Does knx-go supports connecting to a secure tunnel? In case it's supported, can you share an example config?

mobilarte, 2023-01-23:
Currently `knx-go` does not support secure tunneling.

vapourismo, 2023-01-25:
I don't have access to devices that support it, hence I won't be able to implement it. However, I'd happily facilitate if somebody did.

closed 2023-01-25 by vapourismo