vapourismo/knx-go #80: knx-go is writing wrong "source" tag

userwithoutpassword, 2024-03-16:
When Im using DPT 1.005

[[inputs.knx_listener.measurement]]
  name = "Alarm"
  dpt = "1.005"
  addresses = ["0/2/2", # Windalarm (Markise)
               "0/2/3",  # Windalarm (Raffstores)
               "0/2/4"] # Regenalarm
 
knx-go is writing wrong source tag in some situations.

Situation: my sensor is having PA 1.1.50.
At the same time im using ETS6 connected to my KNX-IP Interface (Gira X1) via tunneling connection, having PA 1.1.6

Now i do a "GroupValue_Read" for example on 0/2/2. my Sensor (1.1.50) is responding via GroupValue_Response with its address. But in telegraf im getting 1.1.6 as source tag, but it has to be 1.1.50 instead.

For now other DPTs dont have this problem. First time I notice this but only with DPT 1.005.

I looks like knx-go is taking the PA from the  GroupValue_Read message but the correct address is in the GroupValue_Response message

mobilarte, 2024-03-16:
The individual address in the telegram sent by 1.1.50 is correct. Knx-go does not make up anything, it just reads the content of messages. Maybe telegraf is interpreting GroupValue_Read messages as GroupValue_Response messages.

Please post an ETS trace exhibiting this behavior.

userwithoutpassword, 2024-03-16:
[image]

knx-go is writing both entries to the database (#19 and #20). But only for dpt 1.005
With other dpts knx-go is ignoring the GroupValue_Read Messages

mobilarte, 2024-03-16:
knx-go is not writing anything to any database.  knx-go is not acting differently for DPT_1005 than for DPT_1001, it just unpacks a byte to a bool value. The only difference between the two types is that they have different String() methods.

In your trace (screenshot) I cannot see any message that is a GroupValue_Write or GroupValue_Response originating from any other address than 1.1.51 (The MDT rain sensor). Device 1.1.6 sends a GroupValue_Read to which it gets a legitimate response.

In KNX there is no information in the message about the type of DPT. It is a matter of interpretation by the client of the knx-go library.

Here you have two DPT 1.x messages but you cannot tell which DPT they are

```
write    1.2.9    ->    7/0/0  []byte{0x1}
write    1.2.74   ->    1/4/1  []byte{0x0}
```

In ETS you had to specify that 0/2/4 is a DPT 1.005, otherwise ETS would not know.

userwithoutpassword, 2024-03-16:
Why is knx-go doing this ONLY for 1.005 and not for other DPTs?

mobilarte, 2024-03-16:
I am afraid to have to repeat myself. knx-go is not writing anything to your database, knx-go does not have an interface like your pictures.

The knx_listerner in the plugin for telegraf is responsible for writing the data to the database. Please check the conf file of that plugin for the definition of the DPT.

vapourismo, 2024-03-16:
I don't see anything that is directly caused by knx-go in this issue. Therefore I'm closing it.

I recommend trying to debug this on the telegraph side of things. Perhaps it is not soundly configured.

closed 2024-03-16 by vapourismo