vapourismo/knx-go #81: All DPT1.xxx are boolean

userwithoutpassword, 2024-03-19:
Why are there Strings like "On" "Off"

`// DPT_1001 represents DPT 1.001 (G) / DPT_Switch.
type DPT_1001 bool

func (d DPT_1001) Pack() []byte {
	return []byte{packB1(bool(d))}
}

func (d *DPT_1001) Unpack(data []byte) error {
	return unpackB1(data, (*bool)(d))
}

func (d DPT_1001) Unit() string {
	return ""
}

func (d DPT_1001) String() string {
	if d {
		return "On"
	} else {
		return "Off"
	}
}`

but in influxdb all DPT1.xxx are boolean with values "true" oder "false"

I would expect seeing these Strings filled into the field  "_value"

mobilarte, 2024-03-19:
Datapoint Types are defined in the KNX -> System Specifications -> Internetworking -> Datapoint Types.

The documents can be downloaded [here](https://my.knx.org/en/shop/knx-specifications).

userwithoutpassword, 2024-03-19:
Thats not the question. The Question ist why defining these Strings inside knx-go and dont using them?

mobilarte, 2024-03-19:
I think you completely misunderstand what knx-go does. 

The code that is using knx-go is [here](https://github.com/influxdata/telegraf/blob/master/plugins/inputs/knx_listener/README.md), the KNX Input Plugin used by telegraf.

You should ask them, not here!

userwithoutpassword, 2024-03-19:
ok thx

closed 2024-03-19 by vapourismo